lewapek/sds-dust-sensors-arduino-library #3: Library not working well with hardware serial

mellbratt, 2018-05-17:
Trying to use the library with Adafruit's [Feather m0 LoRa](https://www.adafruit.com/product/3178), a SAMD based board that has several assignable hardware serial ports but doesn't play well with the software serial library. I've tried to make some adaptations but have gotten stuck. Any plans to update the library and accommodate streams that aren't software serial?

lewapek, 2018-05-19:
HardwareSerial support should be merged soon in PR #4

lewapek, 2018-05-19:
Done

closed 2018-05-19 by lewapek

lewapek, 2018-05-20:
Hi,
SoftwareSerial library may not be available for all boards.
I think preprocessor directives are the correct way to make compiler happy however they always pollute the codebase... the challenge is to write as few ```#ifndef ...``` statements as possible, so the code remains clean. 
Feel free to open PR :)

proffalken, 2019-03-17:
@mellbratt if you can put up a working copy of your code in a gist (gist.github.com) then I'd be happy to fork the repo, add you changes, submit the PR, and make sure you get full credit for it?

I'm stuck trying to connect my SDS011 to the SODAQ ExpLoRer and other ESP32-based boards for which SoftwareSerial doesn't exist, and none of the SDS011 libraries that I can find seem to support these boards!

PR #4 (2018-05-19): Adding support for HardwareSerial [fixes #3]
This PR fixes #3 by adding support for HardwareSerial communication
commit 2fd5399e (2018-05-19): Merge pull request #4 from lewapek/feature/hardwareSerial

Adding support for HardwareSerial [fixes #3]