lewapek/sds-dust-sensors-arduino-library #6: Issue when waking up

mvrueden, 2018-08-28:
Hello @lewapek,
thank you very much for your library.

I tried to use it, but run into a weird issue.
If using the continous mode without sleeping everything works fine.
But as soon as I put the sensor into sleep mode and wake it up I get weird responses.

For debugging purposes I implemented my own "library" which just sends commands to the sensor and reads the response. So, putting the sensor to sleep works. Waking it up return seemingly random 5 bytes, for example, when waking the sensor up after a 30 seconds sleep, this is what the sensor responds:

```
0: ad
0: 50
0: 10
0: 5e
0: dd
```
The `0:` is indicating the index. 
As the `AA` is missing, no new message is identified and thus the index is `0` all the time.

From this point on all communication works as specified, e.g. sending another `wake up`, returns the following bytes

```
0: aa
1: c5
2: 6
3: 1
4: 1
5: 0
6: ec
7: e1
8: d5
9: ab
=> Message received: AA C5 6 1 1 0 EC E1 D5 AB 
```

Did you observe a similar behaviour or do you have any idea what might be wrong with my sensor?

lewapek, 2018-08-28:
Hi,
I've checked this problem and it seems that 'continuous mode' and 'wakeup' don't work as expected. Did you encountered similar problem when using query reporting mode?
I will investigate it more deeply later.

mvrueden, 2018-08-29:
My findings are related to `query reporting mode`, as it is easier to debug. I never tested `continous mode`. If you want, I can provide you with my debugging code (for Arduino Nano, though) and how I set up the sensor. 

What I did:
setup()
- query firmware
- set reporting mode to `query`
- set to sleep mode

loop()
- read all serial data until nothing is available anymore
- sleep n seconds
- wakup
- continue at step 1

Everything is returning a valid message. Only waking up sends 5 random wrong bytes. Sending it twice though, will generate at least 1 valid response message.

lewapek, 2018-08-30:
Thanks for your response!
I've implemented double wakeup (#7) to avoid such issue.
It seems that wakeup works ok - sensor fan starts working but sensor sends random bytes or doesn't send anything at all...
You can use wakeup when you want to read and check response or wakeupSingle which may result in incorrect response (however seems to awake sensor).

#7 also handles automatic retries

closed 2018-08-30 by lewapek

lewapek, 2018-08-30:
@mvrueden 
You can now try 1.3.0 release
https://github.com/lewapek/sds-dust-sensors-arduino-library/releases/tag/1.3.0

PR #7 (2018-08-30): Feature/retry failures
double wakeup by default to ensure proper sensor response - this should fix #6 
retry mechanism in case of not available response or invalid head