lewapek/sds-dust-sensors-arduino-library #10: Invalid firmware version

lewapek, 2018-11-18:
Sensor board sometimes returns incorrect firmware version with status OK.
Some of tested sds011 sensors returned 1.0.0 version [year.month.day].

Solution:
version should be checked against possible ranges (at least month and day) and result in different status (not ok - it may require new status)

cyberman54, 2022-02-16:
My board returns with

`ESP_LOGI(TAG, "SDS011 firmware version %s", (sds.queryFirmwareVersion().toString()).c_str());`

this

`SDS011 firmware version Firmware version [year.month.day]: -1.-1.-1`

lewapek, 2022-02-16:
I remember I had the same problem - that's why I created the issue :)
Sometimes it happens - the solution is to retry - you should get the correct version (maybe the sensor needs some time to warm-up)

cyberman54, 2022-02-16:
It seems a `sds.wakeup()` right after opened the serial bus, before sending first command, avoids this issue.

closed 2026-10-14 by lewapek