lewapek/sds-dust-sensors-arduino-library #15: Issue with The Things Uno (LoraWAN Leonardo).

HaViGit, 2019-08-02:
First of all thanks for sharing this library which I use on an Arduino Leonardo or, te be more specific, on a The Things Uno board which is a Leonardo with integrated LoRaWAN.

Using the (unmodified) hardware.Serial.ino demo sketch the first measurement works fine and shows the PM10 and PM25 values in the serial monitor. Every following measurment shows bot PM values as 0.00 (zero). After a reset or boot the same issue is shown:  only the first measurement is correct. The SDS011 sensor seems to be okay because it works fine in another setup with an Arduino Uno.

Is this a known issue or am I doing something wrong and can you advice me how to solve this?

lewapek, 2019-08-02:
Hi,
it is definitely not known issue.
Do you use unmodified version of https://github.com/lewapek/sds-dust-sensors-arduino-library/blob/master/examples/hardwareSerial/hardwareSerial.ino ?
I'm sure I've tested all examples at the time the were implemented - however I've used regular Arduino Leonardo as far as I remember (not "The Things Uno board which is a Leonardo with integrated LoRaWAN").
Also, could you paste here a few lines from serial monitor? The example sketch mentioned above (hardwareSerial.ino) uses sleep/awake command too - do they work as expected?

PS
If anyone using this library with HardwareSerial is reading my comment - it would be grate if you could test https://github.com/lewapek/sds-dust-sensors-arduino-library/blob/master/examples/hardwareSerial/hardwareSerial.ino sketch if it is working properly and give us a note here.

HaViGit, 2019-08-02:
Thanks for your fast and kind reply! I use the unmodified sketch as you mention and I know it sleeps according to the settings , I can hear the fan stop and run as expected. Below you see the output of the serial monitor. Hope to hear from you again.

PM2.5 = 3.30, PM10 = 6.00
pm25: 3.30, pm10: 6.00
Sensor is sleeping
PM2.5 = 0.00, PM10 = 0.00
pm25: 0.00, pm10: 0.00
Sensor is sleeping
PM2.5 = 0.00, PM10 = 0.00
pm25: 0.00, pm10: 0.00
Sensor is sleeping
PM2.5 = 0.00, PM10 = 0.00
pm25: 0.00, pm10: 0.00
Sensor is sleeping
PM2.5 = 0.00, PM10 = 0.00
pm25: 0.00, pm10: 0.00
Sensor is sleeping

lewapek, 2019-08-02:
It seems that the query status is Ok - otherwise, there would be an error printed.

You can also try to use this library with sources and "#define __DEBUG_SDS_DUST_SENSOR__" in SdsDustSensor.h file - then all raw bytes send/received to/from a sensor will be printed in serial monitor as well as all intermediate failures (in case of automatic retries).

I also hope someone else will test this and inform us here about the results.

HaViGit, 2019-08-03:
I tried to use **#define DEBUG_SDS_DUST_SENSOR** in hardwareSerial sketch but this does, as expected (?), not show any information. The SdsDustSensor sketch as mentioned does not use hardware serial and I don't know how to use and wire the I/O, as far as I know this differs from an Arduino Uno.

Maybe intereresting to mention the FW version is not shown at all when using the hardware serial sketch and the mode is reported as "undefined".

misan, 2019-08-03:
I am afraid that some persistent change has been made to your sensor,
disabling the periodic reporting. If that is the case, you'll need to
restore it to the periodic reporting.

HaViGit, 2019-08-03:
Thanks for your suggestion. When I use the sensor with Lewapek's library demo's on an Arduino Uno the FW date is shown as expected. So I do no think it is a sensor issue.

misan, 2019-08-03:
I mentioned it because one of my students had exactly that issue in the
past. If you connect the sensor to the Arduino Serial Monitor and see
nothing is being transmitted then that might be your problem too.
Otherwise, you will need to keep on searching.

lewapek, 2019-08-03:
@misan
thanks for your response

@HaViGit 
`#define DEBUG_SDS_DUST_SENSOR` is resolved at compile time so you need to download and compile this library sources with debug enabled. Then you should see more output in serial monitor

HaViGit, 2019-08-04:
Sorry for the misunderstanding, now I have added the line to the SdsDustSensor.h file as follows:

#ifndef __SDS_DUST_SENSOR_H__
#define __SDS_DUST_SENSOR_H__
#define DEBUG_SDS_DUST_SENSOR

After recompiling and uploading the unmodified hardwareSerial sketch there is no debug info shown in serial monitor. I suppose I'm doing something wrong?

HaViGit, 2019-08-04:
After adding some delays (not sure if they are all needed)  and removing the lines printing the firmware version and modus (mostly dit not work correct) the sketch below now runs about 2 hours without missing a measurement so far.

> #include "SdsDustSensor.h"

SdsDustSensor sds(Serial1);
long interval = 300000;

void setup() {
  delay(2000);
  Serial.begin(115200);
  delay(1000);
  Serial.println("Start using SdsDustSensor.h");
  sds.begin();
  delay(1000);
}

void loop() {
  sds.wakeup();
  delay(1000);
  Serial.println("Fan ON for 30 seconds");
  delay(30000);
  PmResult pm = sds.queryPm();
  delay(1000);
  if (pm.isOk()) {
    Serial.print("PM2.5 = ");
    Serial.print(pm.pm25);
    Serial.print(", PM10 = ");
    Serial.println(pm.pm10);
  } else {
    Serial.print("Could not read values from sensor, reason: ");
    Serial.println(pm.statusToString());
  }
  WorkingStateResult state = sds.sleep();
  delay(1000);
  if (state.isWorking()) {
    Serial.println("Problem with sleeping the sensor.");
  } else {
    Serial.print("Sensor sleeps for ");
    Serial.print(interval / 1000);
    Serial.println(" seconds");
    delay(interval);
  }
}

HaViGit, 2019-08-06:
Do you know if my firmware 2016 version is outdated?

lewapek, 2019-08-08:
Maybe this is somehow connected with firmware version - I'm not sure. As far as I remember I've tested sensors with firmware from 2017 and 2018.
This library is implementation of "Laser Dust Sensor Control Protocol V1.3" - maybe older firmware uses another version of the protocol

HaViGit, 2019-08-08:
Thanks again!

closed 2019-09-16 by lewapek