lewapek/sds-dust-sensors-arduino-library #16: Incompatibility with SD.h

tjiagoM, 2019-08-07:
Hello,

Thanks for your repository, I was able to use it to read from my sensor.

I wanted to save those readings to the microSD card attached to Arduino. However, by just including in the code `#include <SD.h>`, the Arduino simply doesn't work anymore (I cannot even read anything from the Serial, I'm guessing that's because setup() is executed).

Can you give any help trying to debug this, please? I'm guessing some part is shared across these 2 libraries... Have you ever had this problem? How do you usually save the readings from the sensor?

Thanks.

lewapek, 2019-08-08:
Hi,
maybe some other part of code also uses SoftwareSerial and messes up the data on it. Maybe try using other pins than 10,11.
I've never used `SD.h` with this library - I usually send http requests to another devices.

tjiagoM, 2019-08-08:
Thanks, I've tried other pins than 10/11 and it didn't work either. It really seems like they both use the same things in a way that is not compatible... And it was just by calling the SdsDustSensor constructor :/ 

I've tried to look to both source codes but I'm very recent to Arduino so I don't have a lot of experience. It's a pity you never used SD.h, do you have any other recommendation for me to try to find a solution?

I have to say though it would be nice if your tool would be compatible with Arduino's SD library :D 

Thanks anyway!

lewapek, 2019-08-08:
I'm guessing a little bit but maybe it is connected with #14 - if the same change is made to Arduino's SoftwaresSerial
If so you can try using older version of `SD.h` explicitly (which may use compatible SoftwareSerial)

If you overcome the difficulties with `SD.h` and the problem will be connected with this library, I'd be glad if you open merge request

lewapek, 2019-08-08:
I've checked a moment ago and it seems that Arduino's SoftwareSerial didn't change. However there may be other incompatibilities, so trying different `SD.h` versions can hopefully bring us closer to the solution.

closed 2021-09-08 by lewapek

Elspider, 2024-03-19:
Same problem
I am using an Arduino Uno. The serial monitor print the data of the SDS011 until i just import 

> SD.h

Then It start printing nothing, even without any function.
I have used the 1.0 version of SD, i tried to rename the directory and the file in the library to solve ambiguity but i couldn't fix the problem.

lewapek, 2024-04-15:
hmm - looks like some incompatibility as mentioned in the discussion above  
you can try use the library from the sources and remove all occurrences / usages of `SoftwareSerial` - maybe than helps