lewapek/sds-dust-sensors-arduino-library #21: Consistent high readings using this code

thomasmeeus, 2020-06-06:
Hi,

i'm just experimenting a bit with an sds011 sensor and this code. First I hooked up the sds to the accompanying usb adaptor together with this tool: https://github.com/karlchen86/SDS011 and got these readings:

```
Type: set mode: query
PM2.5: 2.00, PM10: 3.00
PM2.5: 1.90, PM10: 2.80
PM2.5: 1.80, PM10: 2.70
PM2.5: 1.70, PM10: 3.10
PM2.5: 1.70, PM10: 3.00
PM2.5: 1.60, PM10: 3.10
´´´
Which seem quite ok since I'm testing indoors. 
However while testing this code, I'm getting consistently way higher readings:

´´´
Using example sketch:

16:18:08.071 -> pm25: 88.00, pm10: 95.00
16:18:09.035 -> PM2.5 = 91.10, PM10 = 98.30
16:18:09.068 -> pm25: 91.10, pm10: 98.30
16:18:10.031 -> PM2.5 = 92.40, PM10 = 99.70
16:18:10.065 -> pm25: 92.40, pm10: 99.70
16:18:11.027 -> PM2.5 = 93.60, PM10 = 101.00
```

I've got 2 SDS011 sensors and both share the same behaviour. Could you maybe explain the difference? Do you maybe use a different calculation method?

lewapek, 2020-06-06:
The method I use:
https://github.com/lewapek/sds-dust-sensors-arduino-library/blob/master/src/SdsDustSensorResults.h#L76

The method https://github.com/karlchen86/SDS011 uses:
https://github.com/karlchen86/SDS011/blob/master/sds011.c#L23

Both seems to be the same. Maybe `rawBytes` aren't correct in one of the solutions?
You can print `rawBytes` by defining __DEBUG_SDS_DUST_SENSOR__ (you need to compile the sources yourself)

thomasmeeus, 2020-06-07:
I'm getting this kind of response when setting the debug param. Does this look alright at first sight? I don't know if all the retries are normal. I'll try to get the same output of the other library and try to compare them.
```
12:22:25.501 -> Working period: continuous
12:22:25.514 -> | <- read bytes with no more available
12:22:25.514 -> Retry #0 due to not available response
12:22:25.514 -> | <- read bytes with no more available
12:22:25.514 -> Retry #1 due to not available response
12:22:25.675 -> | <- read bytes with no more available
12:22:25.675 -> Retry #2 due to not available response
12:22:25.675 -> |AA|C0|7E|3|C2|3|3E|B2|36|AB| <- read bytes with success
12:22:25.695 -> PM2.5 = 89.40, PM10 = 96.20
12:22:25.728 -> pm25: 89.40, pm10: 96.20
12:22:26.127 -> | <- read bytes with no more available
12:22:26.160 -> Retry #0 due to not available response
12:22:26.193 -> | <- read bytes with no more available
12:22:26.260 -> Retry #1 due to not available response
12:22:26.293 -> | <- read bytes with no more available
12:22:26.326 -> Retry #2 due to not available response
12:22:26.359 -> | <- read bytes with no more available
12:22:26.426 -> Retry #3 due to not available response
12:22:26.459 -> | <- read bytes with no more available
12:22:26.492 -> Retry #4 due to not available response
12:22:26.525 -> | <- read bytes with no more available
12:22:26.592 -> Retry #5 due to not available response
12:22:26.625 -> |AA|C0|A5|3|ED|3|3E|B2|88|AB| <- read bytes with success
12:22:26.691 -> PM2.5 = 93.30, PM10 = 100.50
12:22:26.725 -> pm25: 93.30, pm10: 100.50
12:22:27.090 -> | <- read bytes with no more available
12:22:27.157 -> Retry #0 due to not available response
12:22:27.268 -> | <- read bytes with no more available
12:22:27.268 -> Retry #1 due to not available response
12:22:27.268 -> | <- read bytes with no more available
12:22:27.299 -> Retry #2 due to not available response
12:22:27.356 -> | <- read bytes with no more available
12:22:27.389 -> Retry #3 due to not available response
12:22:27.455 -> | <- read bytes with no more available
12:22:27.488 -> Retry #4 due to not available response
12:22:27.521 -> | <- read bytes with no more available
12:22:27.554 -> Retry #5 due to not available response
12:22:27.621 -> |AA|C0|C3|3|D|4|3E|B2|C7|AB| <- read bytes with success
12:22:27.654 -> PM2.5 = 96.30, PM10 = 103.70
12:22:27.687 -> pm25: 96.30, pm10: 103.70
12:22:28.086 -> | <- read bytes with no more available
12:22:28.119 -> Retry #0 due to not available response
12:22:28.185 -> | <- read bytes with no more available
12:22:28.219 -> Retry #1 due to not available response
12:22:28.252 -> | <- read bytes with no more available
12:22:28.285 -> Retry #2 due to not available response
12:22:28.351 -> | <- read bytes with no more available
12:22:28.385 -> Retry #3 due to not available response
12:22:28.418 -> | <- read bytes with no more available
12:22:28.451 -> Retry #4 due to not available response
12:22:28.517 -> | <- read bytes with no more available
12:22:28.551 -> Retry #5 due to not available response
12:22:28.584 -> | <- read bytes with no more available
12:22:28.650 -> Retry #6 due to not available response
12:22:28.683 -> |AA|C0|D9|3|25|4|3E|B2|F5|AB| <- read bytes with success
12:22:28.750 -> PM2.5 = 98.50, PM10 = 106.10
12:22:28.783 -> pm25: 98.50, pm10: 106.10
```

lewapek, 2020-06-11:
Results are consistent with bytes shown above, 3rd and 5th bytes have values 3 or 4. It means that PMs are >= 256*3/10 = 76.8.

Retries mean that stream (from sensor) doesn't contain data. When data appear, they are processed correctly (I guess - no other errors, InvalidHead, InvalidResponseId...)

Unfortunately I don't have much time to test and compare these 2 libraries.
However if anyone can check them (my library and https://github.com/karlchen86/SDS011) it would be great to write your observations here.

taunoe, 2020-07-10:
It seems that https://github.com/karlchen86/SDS011 library does not exist anymore or it is not public anymore.

lewapek, 2020-07-10:
Thanks
I think we can close close this issue now

closed 2020-07-10 by lewapek