lewapek/sds-dust-sensors-arduino-library #23: Compilation fails with SoftwareSerial.h not found. Library.json version doesn't match tag 1.4.0?

sankara, 2020-08-02:
When I try to install this library using the library manager, it fails saying "SoftwareSerial.h" is not found. I looked into the codebase and I clearly see that you have #ifndef statements to validate if SoftwareSerial is required before including it. So I opened up the downloaded code and found that it was different from what I saw on Github. I was able to workaround the issue then by manually downloading the codebase and installing it as zip. 

I did further research and found you have specified the library version as 1.3.2 in library.json whereas the changes that has the #ifndef statements were made after v1.3.2 tag was released. The latest tag seems to be 1.4.0. I'll follow this up with a PR to change the library version (assuming my diagnosis is correct).

sankara, 2020-08-02:
Here's the PR: #24

lewapek, 2020-08-02:
PR merged
Thanks!

closed 2020-08-02 by lewapek

sankara, 2020-08-03:
@lewapek You may have to create a new tag/release for the indexer to pick it up.

lewapek, 2020-08-04:
You're right
There is now 1.4.1 tag and release with proper versions included