lewapek/sds-dust-sensors-arduino-library #27: Compilation fails on Arduino Nano 33 BLE (Sense)

mafavaron, 2021-03-25:
Hello, and first of all thanks for your great and useful library!

By compiling code containing reference to "SdsDustSensor.h" on Arduino IDE 1.8.42.0 using card "Arduino Nano 33 BLE" I get the following error:

............\Documents\Arduino\libraries\Nova_Fitness_Sds_dust_sensors_library\src/SdsDustSensorResults.h:6:10: fatal error: SoftwareSerial.h: No such file or directory

 #include <SoftwareSerial.h>

I've checked the code and in "Serials.h" found the lines

#ifndef ARDUINO_SAMD_VARIANT_COMPLIANCE // there is no SoftwareSerial available (needed) on SAMD boards.
#include <SoftwareSerial.h>
#endif
#include <HardwareSerial.h>

My suspect is, the Nano 33 BLE is a different category than ARDUINO_SAMD, and as with ARDUINO_SAMD the "SoftwareSerial.h" library is missing.

lewapek, 2021-09-08:
Thanks!
You're right - `#ifndef` needs to be modified to exclude `SoftwareSerial` for that board too.
@mafavaron 
If you managed to solve it, please open PR with the change :)

Ernstbert, 2021-11-12:
> Hello, and first of all thanks for your great and useful library!
>
> By compiling code containing reference to "SdsDustSensor.h" on Arduino IDE 1.8.42.0 using card "Arduino Nano 33 BLE" I get the [...]

Hello! 
Could you find a way to make this work?! I am also using the Arduino Nano 33 BLE and I would like to use this library for the SDS011 dust sensor.

lewapek, 2021-11-30:
Hi,
I'm not using that board and I didn't test any solution.
In case you it working for that board new PR is more than welcome :)