lewapek/sds-dust-sensors-arduino-library #33: Using hardwareserial crashing in sds.begin() on arduino-espressif32 framework

cyberman54, 2022-02-15:
I am on arduino-espressif32 framework, thus using HardwareSerial.
 I don't get the library run on top of this, it's crashing in sds.begin() with a Guru Meditation error:

```
17:44:31.126 > [I][main.cpp:371] setup(): init fine-dust-sensor
17:44:31.132 > Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
17:44:31.334 >   #0  0x400d9534:0x3ffd09a0 in SdsDustSensor::execute(Command const&) at .pio/libdeps/usb/Nova Fitness Sds dust sensors library/src/SdsDustSensor.h:165
17:44:31.334 > Rebooting...
```

My serial initialization code is:

```
HardwareSerial sds_Serial(2); // use UART #2
SdsDustSensor sds(sds_Serial);

void setup {
  sds_Serial.begin(9600, SERIAL_8N1, 19, 23);
  sds.begin();
  ...
}
```

Any hints or suggestions what's wrong here?

lewapek, 2022-02-15:
hmm - not used it on `arduino-espressif32` - `execute` method fails - where the hardware serial is used (`available()`, `read()` and `write()` methods)

cyberman54, 2022-02-16:
Solved by PR #34

closed 2022-02-16 by cyberman54