lewapek/sds-dust-sensors-arduino-library #35: invalid initialization of non-const reference of type 'HardwareSerial&' , sds(rxPin, txPin);

dr2okevin, 2022-12-30:
When I try to compile the example for a ES32 board, I get this error in the Arduino IDE.

```
Feinstaub:33:31: error: invalid initialization of non-const reference of type 'HardwareSerial&' from an rvalue of type 'HardwareSerial'
 SdsDustSensor sds(rxPin, txPin);
                               ^
In file included from /home/kevin/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/Arduino.h:154:0,
                 from sketch/Feinstaub.ino.cpp:1:
/home/kevin/.arduino15/packages/esp32/hardware/esp32/1.0.6/cores/esp32/HardwareSerial.h:56:5: note:   after user-defined conversion: HardwareSerial::HardwareSerial(int)
     HardwareSerial(int uart_nr);
     ^
In file included from /home/kevin/Arduino/Feinstaub/Feinstaub.ino:2:0:
/home/kevin/Arduino/libraries/Nova_Fitness_Sds_dust_sensors_library/src/SdsDustSensor.h:60:3: note:   initializing argument 1 of 'SdsDustSensor::SdsDustSensor(HardwareSerial&, int, int)'
   SdsDustSensor(HardwareSerial &hardwareSerial,
   ^
```

Line 31-33 are the following:
```C
int rxPin = 21;
int txPin = 22;
SdsDustSensor sds(rxPin, txPin);
```

lewapek, 2023-01-16:
Hi,
Thanks for pointing this out.

I haven't tested (compiled) the library myself on ESP32 boards. If in the future I will use this with ESP32 - I will make required adjustments in the source code. In the meantime, if you overcome the issue yourself, feel free to create a PR

icepick3000, 2024-05-06:
Any news on this issue? I am running into the same issues.